hombach/ioBroker.kostal-piko-ba #252: Feature request - Add Power/Current/Vols per AC Line

GRaptor, 2022-01-30:
I use the version 1.2.1 of the adapter with a Piko 12.
Works w/o like a charme. Thank you for that!

I don't know about the Piko BA, but on the Piko 12 you may have the current, power and volt per AC line as data available as well.
At least, it's shown in the GUI.
So, do you think it's easy to make these parameters (in addition to the already existing Surplus object) available as well?

If I can assist in any way please let me know (of course, we can write in German as well)

Thanks
GRaptor

hombach, 2022-01-30:
Hi Raptor,

keeping conversation in English is welcome. You're welcome, too ;)

Thank's alot for your feedback about a working adapter with Piko 12.

Looking at the feature request - I think you're talking about this grid values: 

_// live values - grid phase 1
const ID_L1GridCurrent                = 67109377;  // in A  -  not implemented
const ID_L1GridVoltage                = 67109378;  // in V  -  not implemented
const ID_L1GridPower                  = 67109379;  // in W  -  not implemented
// live values - grid phase 2
const ID_L2GridCurrent                = 67109633;  // in A  -  not implemented
const ID_L2GridVoltage                = 67109634;  // in V  -  not implemented
const ID_L2GridPower                  = 67109635;  // in W  -  not implemented
// live values - grid phase 3
const ID_L3GridCurrent                = 67109889;  // in A  -  not implemented
const ID_L3GridVoltage                = 67109890;  // in V  -  not implemented
const ID_L3GridPower                  = 67109891;  // in W  -  not implemented_

This is already prepared... so not that big issue, no help needed, besides some testing ;)

Are just the power values needed? Or also V & A?  I think same refresh time than for DC values would be best.

GRaptor, 2022-01-30:
Thanks for the fast reply!
Yes, that's looks good to me! Geat.
For me, power is what I'm looking for. However, bigger is better so to make all three values availble wouldn't hurt. But I really can't think of any usecase to have the volts available.
Agree, no different refresh time settings needed. Keep it simple :-)
For sure, I can test it. After Kostal changed the com board of my Piko, it finally started publishing data :-)

hombach, 2022-01-30:
Looks like changes are done - Voltage is showing up.... for testing amp and power I would need some sun on the panels :)  ...  tomorrow

hombach, 2022-01-31:
Ok looks fine with Piko BA also with some power on the line - release 1.4.0 is published an deployed in iOBroker latest repo. Please test if it works with your PIKO 12, too....

GRaptor, 2022-01-31:
Awesome!
Just installed it. Seems to work. Will see tomorrow.
Only issue: when starting the adapter, it logs an error:

`2022-01-31 21:59:26.098	error	Please verify IP address: 192.168.2.20 !!!`
`2022-01-31 21:59:26.097	error	Error in calling Kostal Piko API: SyntaxError: Unexpected end of JSON input`

besides that it seems to work

hombach, 2022-01-31:
You've seen this message only once? Should be more often in case of an "real" error. Please check.

GRaptor, 2022-01-31:
For now, only if I (re)start the adapter. But in that case always.

hombach, 2022-02-01:
Errr - no i misunderstood the logging  -  It's another point.... I'll optimize the logging, you'll get another release -  let's say this evening

GRaptor, 2022-02-01:
> Ah.... ok - trouble with the daily statistics data... error should repeat every minute.... or whatever you put as timing in there (60000ms is the standard).

I checked on that already. It's not linked to daily report. 
The error really shows up randomly, not linked to the daily report.

One side note: By accident, (means Zufall), my electrician called today. Told me, something is weird with the reporting of my Piko to the Kostal portal. Anyway, in the end Kostal called me, checked online some settings on my Piko by using a Teamviewer connection. Then said, something is wrong, they have to investigate?! They have an issue getting correct data transmitted to their portal.
Locally I have no issues (beside the one discussing here).
Just want to mention it. I have no clue what they identified as they didn't really tell me.
But, maybe our issue here has the same root cause. So don't dig to deep for now in case it's an error of my Piko.

Still, I'm happy to try your updated version :-)

hombach, 2022-02-01:
My idea is that your piko has some trouble when I'm polling about cycle times of the battery... you have none ;) 
There is also the possibility that Piko is reporting the API error we are producing with iOBroker to Kostal.... although I never seen this in my own tests.

hombach, 2022-02-01:
uuhhmmm  -  i changed nothing besides more logging info - so when you're facing the error again, please post the log last time

hombach, 2022-02-01:
looks like I got the error - root cause was bad timing caused by too much requests for the inverter. Please try 1.4.2. Deployment to latest repo is done... should be viewable in Admin soon.

GRaptor, 2022-02-02:
I think good news: 1.4.2 ran whole day w/o any error message.
Great job, thanks!

closed 2022-02-02 by GRaptor