hombach/ioBroker.kostal-piko-ba #301: Support for PIKO 7.0 version 5.41 / question or new feature request?

seebaernd, 2022-04-29:
I am new at github so be patient if i made something wrong.

I tried to use your adapter for my approx. 5 years old KOSTAL PIKO 7.0
This did not work, i think my PIKO 7.0 is not compatible with your adapter. But perhaps it is just my problem.

As far as i guess my PIKO 7.0 only has a built-in webserver to get information as a html-page, see attached screendump. Do you see a chance, to get my old PIKO 7.0 could work with your adapter? Or are there hints to get the existing adapter work with my PIKO?

[image]

As i am no programmer, i only build a small php-script that is frequently called from crontab to set some values of my PIKO 7.0 in iobroker. This is working for me as good workaround.

#!/usr/bin/php
//<?php
   $url = "http://pvserver:passwort@192.168.1.34";  // Hier die Webseite vom Websever eintragen pvserver = user 
   $io_set_G = "http://server2:8087/set/0_userdata.0.Datenpunkte-iobroker.Kostal-Pico-WR.Energie-Gesamt?value=";
   $io_set_T = "http://server2:8087/set/0_userdata.0.Datenpunkte-iobroker.Kostal-Pico-WR.Energie-Tag?value=";
   $io_set_A = "http://server2:8087/set/0_userdata.0.Datenpunkte-iobroker.Kostal-Pico-WR.AC-aktuell?value=";
   $io_set_end = "&ack=true";

   $Ausgabe = file_get_contents("$url", "r");

    //AC-Leistung_Aktuell

   $pos1 = strpos($Ausgabe,"aktuell</td>")+20;
   $pos1 = strpos($Ausgabe,">",$pos1) + 1;
   $pos2 = strpos($Ausgabe,"</td>",$pos1);
   $data = substr($Ausgabe,$pos1,$pos2-$pos1);
   $AC_P_aktuell = (float) $data;

   //Energie_Gesamtertrag

   $pos1 = strpos($Ausgabe,"Gesamtenergie</td>")+20;
   $pos1 = strpos($Ausgabe,">",$pos1) + 1;
   $pos2 = strpos($Ausgabe,"</td>",$pos1);
   $data = substr($Ausgabe,$pos1,$pos2-$pos1);
   $AC_P_gesamt = (float) $data;

   //Energie_Tagesertrag_Aktuell

   $pos1 = strpos($Ausgabe,"Tagesenergie</td>")+20;
   $pos1 = strpos($Ausgabe,">",$pos1) + 1;
   $pos2 = strpos($Ausgabe,"</td>",$pos1);
   $data = substr($Ausgabe,$pos1,$pos2-$pos1);
   $AC_P_tag = (float) $data;

   //Setze Werte in iobroker über die REST-API

   $Ausgabe = file_get_contents("$io_set_G" . $AC_P_gesamt . $io_set_end , "r");
   $Ausgabe = file_get_contents("$io_set_T" . $AC_P_tag . $io_set_end , "r");
   $Ausgabe = file_get_contents("$io_set_A" . $AC_P_aktuell . $io_set_end , "r");

   exit;
?>

My question is: will i have to use my own php-wokraround in the future or is there a chance to use your adapter with my PIKO 7.0?

Thanx for being patient with me.
Bernd

hombach, 2022-05-01:
I would guess that you put something like "http://192.168.1.34/" as IP adress in the config.... YES? -> please try again with only "192.168.1.34"

seebaernd, 2022-05-01:
I receive following messages in my logfile:

kostal-piko-ba.1 | 2022-05-01 21:27:09.960 | error | Please verify IP address: 192.168.1.34 !! (e2)
-- | -- | -- | --
kostal-piko-ba.1 | 2022-05-01 21:27:09.960 | error | Error in calling Kostal Piko API: HTTPError: Response code 401 (Unauthorized)
kostal-piko-ba.1 | 2022-05-01 21:27:09.957 | error | Please verify IP address: 192.168.1.34 !! (e1)
kostal-piko-ba.1 | 2022-05-01 21:27:09.956 | error | Error in calling Kostal Piko API: HTTPError: Response code 401 (Unauthorized)
kostal-piko-ba.1 | 2022-05-01 21:26:59.958 | error | Please verify IP address: 192.168.1.34 !! (e2)
kostal-piko-ba.1 | 2022-05-01 21:26:59.957 | error | Error in calling Kostal Piko API: HTTPError: Response code 401 (Unauthorized)
kostal-piko-ba.1 | 2022-05-01 21:26:59.954 | error | Please verify IP address: 192.168.1.34 !! (e1)
kostal-piko-ba.1 | 2022-05-01 21:26:59.954 | error | Error in calling Kostal Piko API: HTTPError: Response code 401 (Unauthorized)
kostal-piko-ba.1 | 2022-05-01 21:26:49.957 | error | Please verify IP address: 192.168.1.34 !! (e2)
kostal-piko-ba.1 | 2022-05-01 21:26:49.956 | error | Error in calling Kostal Piko API: HTTPError: Response code 401 (Unauthorized)
kostal-piko-ba.1 | 2022-05-01 21:26:49.952 | error | Please verify IP address: 192.168.1.34 !! (e1)
kostal-piko-ba.1 | 2022-05-01 21:26:49.951 | error | Error in calling Kostal Piko API: HTTPError: Response code 401 (Unauthorized)
kostal-piko-ba.1 | 2022-05-01 21:26:46.933 | error | Please verify IP address: 192.168.1.34 !! (e3)
kostal-piko-ba.1 | 2022-05-01 21:26:46.932 | error | Error in calling Piko API: HTTPError: Response code 401 (Unauthorized)
kostal-piko-ba.1 | 2022-05-01 21:26:44.065 | error | Please verify IP address: 192.168.1.34 !! (e4)
kostal-piko-ba.1 | 2022-05-01 21:26:44.064 | error | Error in calling Piko API for total statistics: HTTPError: Response code 401 (Unauthorized)
kostal-piko-ba.1 | 2022-05-01 21:26:43.869 | info | Polltime alltime statistics set to: 200 seconds
kostal-piko-ba.1 | 2022-05-01 21:26:43.868 | info | Polltime daily statistics set to: 60 seconds
kostal-piko-ba.1 | 2022-05-01 21:26:43.868 | info | Polltime set to: 10 seconds

seebaernd, 2022-05-01:
[image]

hombach, 2022-05-01:
Please try "http://pvserver:test@192.168.1.34/api/dxs.json?dxsEntries=16780544&dxsEntries=16779267&dxsEntries=16777984"  in your favorite webbrowser... 
!! pvserver has to be your username; test your password for the inverter...

please send your browser result.... should be like:
"{"dxsEntries":[{"dxsId":16780544,"value":"PIKO 6.0 BA"},{"dxsId":16779267,"value":"06.41"},{"dxsId":16777984,"value":"Piko"}],"session":{"sessionId":0,"roleId":0},"status":{"code":0}}"

seebaernd, 2022-05-01:
[image]

hombach, 2022-05-02:
Hi Bernd, hard to tell without direct access to your system, but as far as I understood you'll need an Kostal UI => 6.11.
There is an update to Version 6.41 for your inverter on the Kostal Homepage. Perhaps also an FW Update could be usefull. 
https://www.kostal-solar-electric.com/de-de/download/archiv/#Solar-Wechselrichter/PIKO%204.2%20-%208.5/Deutschland/Update/

hombach, 2022-05-03:
FYI - according to sentry there are minimum 4 other user with Piko 7.0 using the adapter.... ;)

seebaernd, 2022-05-03:
Thanx for your effort to help me. Looks that my inverter would need an update to work with your adapter. 

But I am not sure, whether I will make these updates - never touch a running system. I dont have any experience with updates an my PIKO. Also I dont have a service passwort for the inverter and i am not sure, whether everything will work fine after the update. And my php-script and workaround with rest-api of iobroker ist pretty working.

Thanx again for your friendly help.

hombach, 2022-05-03:
You're welcome!

closed 2022-05-03 by hombach