hombach/ioBroker.kostal-piko-ba #337: Error in calling Kostal Piko API: SyntaxError: Unexpected end of JSON input

dj-mib, 2022-08-04:
Adapter is working but every day I have several entries of this error in the logfile.

I have a Piko 10 installed
System Info:
Versions:
UI06.41
FW05.60
HW0100
PAR02.70
Artikelnummer10101181
LändereinstellungGermany NSR
 (30.11)

Do you habe any idea why this happens?

hombach, 2022-08-04:
Hello Michael, 

adapter version? nodejs version? js controller version?
or you could send your Piko's IP adress, then I could check sentry logs about this.

Define several entries .... sometimes? Or quite constant every e.g. every 250sec....

dj-mib, 2022-08-05:
Hello Hombach,

Node.js    v12.22.7
Kostal Adapter 1.4.7
JS Adapter 5.7.0 

Pikos IP is 192.168.0.4
I added some logs regarding the adapter.
Polltime is set to: 7.5 seconds, is it too short? 

The error sounds like Piko is not responding correctly.

[kostal.log](https://github.com/hombach/ioBroker.kostal-piko-ba/files/9265784/kostal.log)

dj-mib, 2022-08-05:
Hello again, 
**I found something stupid.** 
Polltime alltime statistics set to: 0.12 seconds
In the Configuration Polltime was set to 120.000. It is now set to 120000.
I don't know how the decimal separator was added but I think this caused a denial of service attack.

I will check if the error still occurs.

Maybe it is useful to define some minimum values to prevent this.

hombach, 2022-08-05:
Yes, sounds as this really could be an active reaction to a "DoS-Attack"... or just IO overload :)
-> will be definitely better with correct numbers, let's see if it is completely fixed.

-> You're right there should be some limits for the values.... I'll check Kostal docu, perhaps something is defined... plus: I'll define that daily stat poll has to be minimum double of standard poll and alltime double of daily.

hombach, 2022-08-05:
Another point... you're restarting the adapter every 24h .... ? Any troubles in continous run? I never restart - expect for updates

hombach, 2022-08-05:
Tried with 100ms poll - I get the same errors.
Also tested the update frequency - looks like Piko internal update is only every 5 seconds

dj-mib, 2022-08-05:
> Another point... you're restarting the adapter every 24h .... ? Any troubles in continous run? I never restart - expect for updates

Daily restart was an attempt to get rid of this error. No other issues with your adapter. **It works fine!** 
After fixing the update frequency there are no more errors in the log.

Validating the update frequency values will reduce configuration failure.
Thank you very much.

hombach, 2022-08-05:
.... use version 1.5.0 - validation of update values implemented.... ;)

closed 2022-08-05 by hombach

dj-mib, 2022-08-06:
Installed and tested.  Invalid values will be overwritten by minimum values.