hombach/ioBroker.kostal-piko-ba #373: Response code 401

freakyfreaky, 2022-10-10:
**Describe the bug**  
I am using a PIKO 5.5 because we our system is now 10 yrs old.
After setting up everything I get an error messag

ERROR:
Error in calling Kostal Piko API: HTTPError: Response code 401 (Unauthorized)

**To Reproduce**  
Steps to reproduce the behavior:
1. Set up Instance
2. Wait for first poll

**Versions:**  
 - Adapter version: 1.5.0
 - Node version: 16.17.1
 - Operating system: raspberry pi & ubuntu 22

hombach, 2022-10-11:
Any logs in debug mode? Or perhaps you send your IP to check in sentry logs...

freakyfreaky, 2022-10-11:
DEBUG LOG

hombach, 2022-10-11:
Hard to tell without direct access to your system, but as far as I understood you'll need an Kostal UI => 6.11.
There is an update to Version 6.41 for your inverter on the Kostal Homepage. Perhaps also an FW Update could be usefull.
Which version of UI and firmware are you using?

freakyfreaky, 2022-10-12:
A very old one. :)
It is a Piko 5.5 with version 3.75
According first Google result there is no update option for this old buddy

Don’t worry too much if there is no way arround.  I will script something as I did with my old PHP setup. I just wanted to give the adapter a try on a non supported version to test it for you. 

[image]

freakyfreaky, 2022-10-12:
[image]

freakyfreaky, 2022-10-12:
[image]

hombach, 2022-10-16:
Are you sure there is no new firmware available? Because of HW version? By a short check in sentry there have been using 14 PIKO 5.5 owners the adapter in the past few days (without error) UI v6.11; v6.20; v6.30 and v6.41

hombach, 2022-10-16:
Looks like there has been v4 released in summer 2011... as initial version. Perhaps new hardware at the same date?https://www.kostal-solar-electric.com/de-de/download/archiv/#Solar-Wechselrichter/PIKO%204.2%20-%208.5/Deutschland/Update/

closed 2022-10-19 by hombach