hombach/ioBroker.kostal-piko-ba #408: Pico 3.0

gernot1978, 2023-01-28:
Hallo
Ich bekome mit meinem Pico 3.0 funktionierende Datenpunkte.Falls Du mehr infos brauchst kann ich Dir gerne mehr dazu berichten.
Habe nur im Log folgende Fehlermeldung: 
<html><body>

kostal-piko-ba.0 | 2023-01-28 14:32:11.836 | error | Error in calling Kostal Piko API: TypeError: Cannot read properties of undefined (reading 'value')
-- | -- | -- | --
kostal-piko-ba.0 | 2023-01-28 14:32:01.835 | error | Please verify IP address: 192.168.193.42 !! (e1)

</body>
</html>

Grüße Gernot

hombach, 2023-01-28:
Hallo Gernot, du könntest die adapter mal in den debug mode versetzen und posten was dann nach "Piko-BA live data 1 updated - Kostal response data:" im log steht.

gernot1978, 2023-01-29:
Hallo
Anbei der Logeintrag:
Piko-BA live data 2 updated - Kostal response data:
{"dxsEntries":[{"dxsId":67109377,"value":0.741577},{"dxsId":67109378,"value":225.054550},{"dxsId":67109379,"value":83.228195},{"dxsId":67109633,"value":null},{"dxsId":67109634,"value":null},{"dxsId":67109635,"value":null},{"dxsId":67109889,"value":null},{"dxsId":67109890,"value":null},{"dxsId":67109891,"value":null}],"session":{"sessionId":0,"roleId":0},"status":{"code":0}}

Viele Grüße
Gernot

hombach, 2023-01-29:
Hallo Gernot,
dein geposteter Log Teil ist aus dem zweiten Abfrageblock "Piko-BA live data 2 up...."
Dein ursprünglicher Fehler und mein Info-Bedarf wäre für Block 1: "Piko-BA live data 1 up...."
Root cause sollte aber die Einphasigkeit des Piko 3 sein. Soweit ich das im Kostal Archiv sehen konnte hat dieser eine AC Phase, und auch einen DC Eingang. (?) Das habe ich bisher nicht abgefangen.
In Kürze ist eine Version 2.1.2 im Latest Repo verfügbar... Mit etwas Glück habe ich die Probleme abgefangen.

gernot1978, 2023-01-31:
> Hi Gernot, mal probiert?

Ja das Log von vorhin war schon mit version2.1.2 immernoch der Fehler

hombach, 2023-01-31:
Sorry, Log von "vorhin"? - habe ich keines gesehen... Das oben stehende ist nicht von der 2.1.2

gernot1978, 2023-02-01:
Guten Morgen
Hatte versucht per Anhang mein Log zu senden.
Ich versuche es nochmal...
Grüße Gernot

":67109378,"value":222.355545},{"dxsId":67109379,"value":10.880394},{"dxsId":67109633,"value":null},{"dxsId":67109634,"value":null},{"dxsId":67109635,"value":null},{"dxsId":67109889,"value":null},{"dxsId":67109890,"value":null},{"dxsId":67109891,"value":null}],"session":{"sessionId":0,"roleId":0},"status":{"code":0}}
2023-01-31 08:07:32.405 - debug: kostal-piko-ba.0 (3158894) Piko-BA daily statistics updated - Kostal response data: {"dxsEntries":[{"dxsId":251659266,"value":7029811554912652692834746368.000000},{"dxsId":251659278,"value":100.000000},{"dxsId":251658754,"value":0.705886},{"dxsId":251659010,"value":98121357226202500169728.000000},{"dxsId":251659279,"value":100.000000}],"session":{"sessionId":0,"roleId":0},"status":{"code":0}}
2023-01-31 08:07:34.163 - error: kostal-piko-ba.0 (3158894) Error in calling Kostal Piko API: TypeError: Cannot read properties of undefined (reading 'value')
2023-01-31 08:07:34.164 - error: kostal-piko-ba.0 (3158894) Please verify IP address: 192.168.193.42 !! (e1)
2023-01-31 08:07:34.166 - debug: kostal-piko-ba.0 (3158894) Piko-BA live data 2 updated - Kostal response data: {"dxsEntries":[{"dxsId":67109377,"value":0.576782},{"dxsId":67109378,"value":222.741119},{"dxsId":67109379,"value":9.369335},{"dxsId":67109633,"value":null},{"dxsId":67109634,"value":null},{"dxsId":67109635,"value":null},{"dxsId":67109889,"value":null},{"dxsId":67109890,"value":null},{"dxsId":67109891,"value":null}],"session":{"sessionId":0,"roleId":0},"status":{"code":0}}
2023-01-31 08:07:44.168 - error: kostal-piko-ba.0 (3158894) Error in calling Kostal Piko API: TypeError: Cannot read properties of undefined (reading 'value')
2023-01-31 08:07:44.168 - error: kostal-piko-ba.0 (3158894) Please verify IP address: 192.168.193.42 !! (e1)
2023-01-31 08:07:44.171 - debug: kostal-piko-ba.0 (3158894) Piko-BA live data 2 updated - Kostal response data: {"dxsEntries":[{"dxsId":67109377,"value":0.584106},{"dxsId":67109378,"value":221.821686},{"dxsId":67109379,"value":9.837902},{"dxsId":67109633,"value":null},{"dxsId":67109634,"value":null},{"dxsId":67109635,"value":null},{"dxsId":67109889,"value":null},{"dxsId":67109890,"value":null},{"dxsId":67109891,"value":null}],"session":{"sessionId":0,"roleId":0},"status":{"code":0}}
2023-01-31 08:07:54.166 - error: kostal-piko-ba.0 (3158894) Error in calling Kostal Piko API: TypeError: Cannot read properties of undefined (reading 'value')
2023-01-31 08:07:54.167 - error: kostal-piko-ba.0 (3158894) Please verify IP address: 192.168.193.42 !! (e1)
2023-01-31 08:07:54.169 - debug: kostal-piko-ba.0 (3158894) Piko-BA live data 2 updated - Kostal response data: {"dxsEntries":[{"dxsId":67109377,"value":0.582275},{"dxsId":67109378,"value":221.347137},{"dxsId":67109379,"value":12.820864},{"dxsId":67109633,"value":null},{"dxsId":67109634,"value":null},{"dxsId":67109635,"value":null},{"dxsId":67109889,"value":null},{"dxsId":67109890,"value":null},{"dxsId":67109891,"value":null}],"session":{"sessionId":0,"roleId":0},"status":{"code":0}}
2023-01-31 08:08:04.170 - error: kostal-piko-ba.0 (3158894) Error in calling Kostal Piko API: TypeError: Cannot read properties of undefined (reading 'value')
2023-01-31 08:08:04.170 - error: kostal-piko-ba.0 (3158894) Please verify IP address: 192.168.193.42 !! (e1)
2023-01-31 08:08:04.173 - debug: kostal-piko-ba.0 (3158894) Piko-BA live data 2 updated - Kostal response data: {"dxsEntries":[{"dxsId":67109377,"value":0.584106},{"dxsId":67109378,"value":221.139526},{"dxsId":67109379,"value":10.471620},{"dxsId":67109633,"value":null},{"dxsId":67109634,"value":null},{"dxsId":67109635,"value":null},{"dxsId":67109889,"value":null},{"dxsId":67109890,"value":null},{"dxsId":67109891,"value":null}],"session":{"sessionId":0,"roleId":0},"status":{"code":0}}
2023-01-31 08:08:14.173 - error: kostal-piko-ba.0 (3158894) Error in calling Kostal Piko API: TypeError: Cannot read properties of undefined (reading 'value')
2023-01-31 08:08:14.174 - error: kostal-piko-ba.0 (3158894) Please verify IP address: 192.168.193.42 !! (e1)
2023-01-31 08:08:14.176 - debug: kostal-piko-ba.0 (3158894) Piko-BA live data 2 updated - Kostal response data: {"dxsEntries":[{"dxsId":67109377,"value":0.582275},{"dxsId":67109378,"value":221.821686},{"dxsId":67109379,"value":12.677057},{"dxsId":67109633,"value":null},{"dxsId":67109634,"value":null},{"dxsId":67109635,"value":null},{"dxsId":67109889,"value":null},{"dxsId":67109890,"value":null},{"dxsId":67109891,"value":null}],"session":{"sessionId":0,"roleId":0},"status":{"code":0}}
2023-01-31 08:08:24.172 - error: kostal-piko-ba.0 (3158894) Error in calling Kostal Piko API: TypeError: Cannot read properties of undefined (reading 'value')
2023-01-31 08:08:24.172 - error: kostal-piko-ba.0 (3158894) Please verify IP address: 192.168.193.42 !! (e1)
2023-01-31 08:08:24.175 - debug: kostal-piko-ba.0 (3158894) Piko-BA live data 2 updated - Kostal response data: {"dxsEntries":[{"dxsId":67109377,"value":0.582275},{"dxsId":67109378,"value":222.088623},{"dxsId":67109379,"value":11.631633},{"dxsId":67109633,"value":null},{"dxsId":67109634,"value":null},{"dxsId":67109635,"value":null},{"dxsId":67109889,"value":null},{"dxsId":67109890,"value":null},{"dxsId":67109891,"value":null}],"session":{"sessionId":0,"roleId":0},"status":{"code":0}}
[... cut ...] tatus":{"code":0}}
2023-01-31 08:19:04.291 - error: kostal-piko-ba.0 (3158894) Error in calling Kostal Piko API: TypeError: Cannot read properties of undefined (reading 'value')
2023-01-31 08:19:04.291 - error: kostal-piko-ba.0 (3158894) Please verify IP address: 192.168.193.42 !! (e1)
2023-01-31 08:19:04.294 - debug: kostal-piko-ba.0 (3158894) Piko-BA live data 2 updated - Kostal response data: {"dxsEntries":[{"dxsId":67109377,"value":0.000000},{"dxsId":67109378,"value":222.770782},{"dxsId":671

hombach, 2023-02-03:
Hallo Gernot, 

ich werde noch nicht ganz schlau mit dem Piko 3.0. Ich habe nun die 2.1.3 in den latest repo gestellt um die debug Möglichkeiten zu verbessern. 
Würde dich bitte noch einmal eine Stück Debug Log zu schicken.
Der 3.0er Piko hat an irgendeiner Stelle weniger Info Möglichkeiten wie die anderen Typen (außer der Einphasigkeit) die ich noch nicht verstehe.

gernot1978, 2023-02-03:
Hi
Klar kann ich machen...kannst du mir sagen wie ich an die repo komme? so
fit bin ich noch nicht im EDV universum....
Hab momentan nur Stable und Beta....
Grüße Gernot

hombach, 2023-02-03:
Ich meinte die beta - sorry falsch ausgedrückt.

Frage am Rande: Habe eben das Auslesen der drei einzelnen Phasen des Hausverbrauchs implementiert (V 2.2.0). Wie ist das beim 3.0 Piko? Wird da auch ein 3 Phasiger Sensor verwendet?

gernot1978, 2023-02-05:
Habe gerade neu installiert.
Sollte im Log mit drin sein.
Wenn Du magst können wir zusammen mal über TeamView auf den Pico schauen....
Grüße Gernot

hombach, 2023-02-05:
Hi Gernot, Log hier anhängen hat nicht geklappt...

gernot1978, 2023-02-07:
Guten Morgen....
Isz bei dir was angekommen? Hier mal 2 zeilen zum Test....

kostal-piko-ba.0
2023-02-07 06:44:37.875 debug Piko-BA lifetime statistics updated - Kostal
response data:
{"dxsEntries":[{"dxsId":251659265,"value":417002191257096639678983887126528.000000},{"dxsId":251659280,"value":100.000000},{"dxsId":251658753,"value":23183.013672},{"dxsId":251659009,"value":56387607847868770333718044737536.000000},{"dxsId":251659281,"value":100.000000},{"dxsId":251658496,"value":27130},{"dxsId":33556228,"value":0}],"session":{"sessionId":0,"roleId":0},"status":{"code":0}}

kostal-piko-ba.0
2023-02-07 06:44:37.825 debug Piko-BA daily statistics update - Kostal
response data:
{"dxsEntries":[{"dxsId":251659266,"value":0.000000},{"dxsId":251659278,"value":0.000000},{"dxsId":251658754,"value":0.000000},{"dxsId":251659010,"value":0.000000},{"dxsId":251659279,"value":0.000000}],"session":{"sessionId":0,"roleId":0},"status":{"code":0}}

kostal-piko-ba.0
2023-02-07 06:44:33.117 debug Piko-BA live data 2 update - Kostal response
data:
{"dxsEntries":[{"dxsId":67109377,"value":0.000000},{"dxsId":67109378,"value":0.000000},{"dxsId":67109379,"value":0.000000},{"dxsId":67109633,"value":0.000000},{"dxsId":67109634,"value":0.000000},{"dxsId":67109635,"value":0.000000},{"dxsId":67109889,"value":0.000000},{"dxsId":67109890,"value":0.000000},{"dxsId":67109891,"value":0.000000}],"session":{"sessionId":0,"roleId":0},"status":{"code":0}}

kostal-piko-ba.0
2023-02-07 06:44:33.116 error Please verify IP address: 192.168.193.42 !!
(e1)

kostal-piko-ba.0
2023-02-07 06:44:33.115 error Error in calling Kostal Piko API: TypeError:
Cannot read properties of undefined (reading 'value')

kostal-piko-ba.0
2023-02-07 06:44:33.114 debug Piko-BA live data 1 update - Kostal response
data:
{"dxsEntries":[{"dxsId":33556736,"value":0.000000},{"dxsId":67109120,"value":0.000000},{"dxsId":33555203,"value":0.000000},{"dxsId":33555201,"value":0.000000},{"dxsId":33555202,"value":0.000000},{"dxsId":33555459,"value":0.000000},{"dxsId":33555457,"value":0.000000},{"dxsId":33555458,"value":0.000000},{"dxsId":33555715,"value":0.000000},{"dxsId":33555713,"value":0.000000},{"dxsId":33555714,"value":0.000000},{"dxsId":83888128,"value":0.000000},{"dxsId":83887872,"value":0.000000},{"dxsId":16780032,"value":0},{"dxsId":33556226,"value":0.000000},{"dxsId":33556227,"value":0.000000},{"dxsId":33556229,"value":0.000000},{"dxsId":33556238,"value":0.000000},{"dxsId":33556230,"value":0}],"session":{"sessionId":0,"roleId":0},"status":{"code":1}}

kostal-piko-ba.0
2023-02-07 06:44:23.121 debug Piko-BA live data 2 update - Kostal response
data:
{"dxsEntries":[{"dxsId":67109377,"value":0.000000},{"dxsId":67109378,"value":0.000000},{"dxsId":67109379,"value":0.000000},{"dxsId":67109633,"value":0.000000},{"dxsId":67109634,"value":0.000000},{"dxsId":67109635,"value":0.000000},{"dxsId":67109889,"value":0.000000},{"dxsId":67109890,"value":0.000000},{"dxsId":67109891,"value":0.000000}],"session":{"sessionId":0,"roleId":0},"status":{"code":0}}

Grüße Gernot

Komisch… hier im Mailverlauf ist es dran….

hombach, 2023-02-07:
Schönen Abend, 

jetzt sehe ich den Fehler. Dein Piko 3 liefert in einem der Datenpakete einen Block weniger zurück. Hier fehlt die Grid Limitation. Verstehe ich eigentlich nicht warum, denn das hat nichts mit 1 oder 3 phasigem Umrichter zu tun. Auch nicht mit Batterie oder nicht. Es geht um die Abregelung der Dach-Peak-Leistung... ich gehe jetzt davon aus, dass dort bei dir eine Abregelung definiert ist und der Umrichter statt 100% einfach gar nichts meldet.

hombach, 2023-02-07:
Das sollte jetzt aber passen... Version 2.2.1 taucht in den nächsten Stunden im Beta Repo auf. Denke du kannst auch gleich auf Info statt Debug gehen.

hombach, 2023-02-13:
Hallo Gernot, konntest du die V2.2.1 testen? Kommen noch Fehler in den Logs?

gernot1978, 2023-02-13:
Hallo
Ich sag mal jein… wollte auf proxmox umziehen und stecke gerade bis zum
Hals in der Sch….
Aber was ich zuletzt sehen konnte kommen noch Fehler. Ich versuche im Lauf
der Woche wieder verlässliche log zu schicken…
Grüße Gernot

gernot1978, 2023-02-13:
Habe wieder fast alles am laufen... Anbei ein Logauszug
2023-02-13 22:27:20.303 - error: kostal-piko-ba.0 (1386) Please verify IP
address: 192.168.193.42 !! (e1)
2023-02-13 22:27:30.304 - error: kostal-piko-ba.0 (1386) Error in calling
Kostal Piko API: TypeError: Cannot read properties of undefined (reading
'value')
2023-02-13 22:27:30.304 - error: kostal-piko-ba.0 (1386) Please verify IP
address: 192.168.193.42 !! (e1)
2023-02-13 22:27:40.305 - error: kostal-piko-ba.0 (1386) Error in calling
Kostal Piko API: TypeError: Cannot read properties of undefined (reading
'value')
2023-02-13 22:27:40.305 - error: kostal-piko-ba.0 (1386) Please verify IP
address: 192.168.193.42 !! (e1)
2023-02-13 22:27:50.308 - error: kostal-piko-ba.0 (1386) Error in calling
Kostal Piko API: TypeError: Cannot read properties of undefined (reading
'value')
2023-02-13 22:27:50.308 - error: kostal-piko-ba.0 (1386) Please verify IP
address: 192.168.193.42 !! (e1)
2023-02-13 22:28:00.308 - error: kostal-piko-ba.0 (1386) Error in calling
Kostal Piko API: TypeError: Cannot read properties of undefined (reading
'value')
2023-02-13 22:28:00.308 - error: kostal-piko-ba.0 (1386) Please verify IP
address: 192.168.193.42 !! (e1)
2023-02-13 22:28:10.309 - error: kostal-piko-ba.0 (1386) Error in calling
Kostal Piko API: TypeError: Cannot read properties of undefined (reading
'value')
2023-02-13 22:28:10.309 - error: kostal-piko-ba.0 (1386) Please verify IP
address: 192.168.193.42 !! (e1)
2023-02-13 22:28:12.255 - info: host.iobroker stopInstance
system.adapter.kostal-piko-ba.0 (force=false, process=true)
2023-02-13 22:28:12.258 - info: kostal-piko-ba.0 (1386) Got terminate
signal TERMINATE_YOURSELF
2023-02-13 22:28:12.259 - info: kostal-piko-ba.0 (1386) Adapter
Kostal-Piko-BA cleaned up everything...
2023-02-13 22:28:12.259 - info: kostal-piko-ba.0 (1386) terminating
2023-02-13 22:28:12.259 - info: kostal-piko-ba.0 (1386) Terminated
(ADAPTER_REQUESTED_TERMINATION): Without reason
2023-02-13 22:28:12.301 - info: host.iobroker stopInstance
system.adapter.kostal-piko-ba.0 send kill signal
2023-02-13 22:28:12.841 - info: host.iobroker instance
system.adapter.kostal-piko-ba.0 terminated with code 11
(ADAPTER_REQUESTED_TERMINATION)
2023-02-13 22:28:15.400 - info: host.iobroker instance
system.adapter.kostal-piko-ba.0 started with pid 2249
2023-02-13 22:28:15.806 - debug: kostal-piko-ba.0 (2249) Redis Objects: Use
Redis connection: 127.0.0.1:9001
2023-02-13 22:28:15.837 - debug: kostal-piko-ba.0 (2249) Objects client
ready ... initialize now
2023-02-13 22:28:15.838 - debug: kostal-piko-ba.0 (2249) Objects create
System PubSub Client
2023-02-13 22:28:15.838 - debug: kostal-piko-ba.0 (2249) Objects create
User PubSub Client
2023-02-13 22:28:15.898 - debug: kostal-piko-ba.0 (2249) Objects client
initialize lua scripts
2023-02-13 22:28:15.900 - debug: kostal-piko-ba.0 (2249) Objects connected
to redis: 127.0.0.1:9001
2023-02-13 22:28:15.928 - debug: kostal-piko-ba.0 (2249) Redis States: Use
Redis connection: 127.0.0.1:9000
2023-02-13 22:28:15.949 - debug: kostal-piko-ba.0 (2249) States create
System PubSub Client
2023-02-13 22:28:15.949 - debug: kostal-piko-ba.0 (2249) States create User
PubSub Client
2023-02-13 22:28:16.030 - debug: kostal-piko-ba.0 (2249) States connected
to redis: 127.0.0.1:9000
2023-02-13 22:28:16.166 - debug: kostal-piko-ba.0 (2249) Plugin sentry
Initialize Plugin (enabled=true)
2023-02-13 22:28:16.416 - info: kostal-piko-ba.0 (2249) starting. Version
2.2.1 in /opt/iobroker/node_modules/iobroker.kostal-piko-ba, node:
v16.19.0, js-controller: 4.0.24
2023-02-13 22:28:16.430 - info: kostal-piko-ba.0 (2249) IP address found in
[... 21 lines cut ...]
{"dxsEntries":[{"dxsId":251659266,"value":13063102297150725421015279557672960.000000},{"dxsId":251659278,"value":100.000000},{"dxsId":251658754,"value":13711.394531},{"dxsId":251659010,"value":0.000000},{"dxsId":251659279,"value":0.000000}],"session":{"sessionId":0,"roleId":0},"status":{"code":0}}
2023-02-13 22:28:27.473 - debug: kostal-piko-ba.0 (2249) Initial ReadPiko
done
2023-02-13 22:28:27.518 - debug: kostal-piko-ba.0 (2249) Piko-BA live data
1 update - Kostal response data:
{"dxsEntries":[{"dxsId":33556736,"value":0.000000},{"dxsId":67109120,"value":0.000000},{"dxsId":33555203,"value":0.000000},{"dxsId":33555201,"value":0.000000},{"dxsId":33555202,"value":0.000000},{"dxsId":33555459,"value":0.000000},{"dxsId":33555457,"value":0.000000},{"dxsId":33555458,"value":0.000000},{"dxsId":33555715,"value":0.000000},{"dxsId":33555713,"value":0.000000},{"dxsId":33555714,"value":0.000000},{"dxsId":83888128,"value":0.000000},{"dxsId":83887872,"value":0.000000},{"dxsId":16780032,"value":0},{"dxsId":33556226,"value":0.000000},{"dxsId":33556227,"value":0.000000},{"dxsId":33556229,"value":0.000000},{"dxsId":33556238,"value":0.000000},{"dxsId":33556230,"value":0}],"session":{"sessionId":0,"roleId":0},"status":{"code":1}}
2023-02-13 22:28:27.519 - error: kostal-piko-ba.0 (2249) Error in calling
Kostal Piko API: TypeError: Cannot read properties of undefined (reading
'value')
2023-02-13 22:28:27.519 - error: kostal-piko-ba.0 (2249) Please verify IP
address: 192.168.193.42 !! (e1)
2023-02-13 22:28:27.527 - debug: kostal-piko-ba.0 (2249) Piko-BA live data
2 update - Kostal response data:
{"dxsEntries":[{"dxsId":67109377,"value":0.000000},{"dxsId":67109378,"value":0.000000},{"dxsId":67109379,"value":0.000000},{"dxsId":67109633,"value":0.000000},{"dxsId":67109634,"value":0.000000},{"dxsId":67109635,"va [... cut ...]
{"dxsEntries":[{"dxsId":67109377,"value":0.000000},{"dxsId":67109378,"value":0.000000},{"dxsId":67109379,"value":0.000000},{"dxsId":67109633,"value":0.000000},{"dxsId":67109634,"value":0.000000},{"dxsId":67109635,"value":0.000000},{"dxsId":67109889,"value":0.000000},{"dxsId":67109890,"value":0.000000},{"dxsId":67109891,"value":0.000000},{"dxsId":83887106,"value":0.000000},{"dxsId":83887362,"value":0.000000},{"dxsId":83887618,"value":0.000000}],"session":{"sessionId":0,"roleId":0},"status":{"code":0}}

hombach, 2023-02-15:
ok.... da war ich nicht gut genug - Neuer Versuch mit V2.2.2.

gernot1978, 2023-02-16:
Hallo
Hier nochmal ein Log für Dich....
kostal-piko-ba.0
2023-02-16 18:43:08.761 debug Piko-BA live data 2 update - Kostal response
data:
{"dxsEntries":[{"dxsId":67109377,"value":0.000000},{"dxsId":67109378,"value":0.000000},{"dxsId":67109379,"value":0.000000},{"dxsId":67109633,"value":0.000000},{"dxsId":67109634,"value":0.000000},{"dxsId":67109635,"value":0.000000},{"dxsId":67109889,"value":0.000000},{"dxsId":67109890,"value":0.000000},{"dxsId":67109891,"value":0.000000},{"dxsId":83887106,"value":0.000000},{"dxsId":83887362,"value":0.000000},{"dxsId":83887618,"value":0.000000}],"session":{"sessionId":0,"roleId":0},"status":{"code":0}}

kostal-piko-ba.0
2023-02-16 18:43:08.759 debug Piko-BA live data 1 update - Kostal response
data:
{"dxsEntries":[{"dxsId":33556736,"value":0.000000},{"dxsId":67109120,"value":0.000000},{"dxsId":33555203,"value":0.000000},{"dxsId":33555201,"value":0.000000},{"dxsId":33555202,"value":0.000000},{"dxsId":33555459,"value":0.000000},{"dxsId":33555457,"value":0.000000},{"dxsId":33555458,"value":0.000000},{"dxsId":33555715,"value":0.000000},{"dxsId":33555713,"value":0.000000},{"dxsId":33555714,"value":0.000000},{"dxsId":83888128,"value":0.000000},{"dxsId":83887872,"value":0.000000},{"dxsId":16780032,"value":0},{"dxsId":33556226,"value":0.000000},{"dxsId":33556227,"value":0.000000},{"dxsId":33556229,"value":0.000000},{"dxsId":33556238,"value":0.000000},{"dxsId":33556230,"value":0}],"session":{"sessionId":0,"roleId":0},"status":{"code":1}}

kostal-piko-ba.0
2023-02-16 18:43:05.962 debug Piko-BA daily statistics update - Kostal
response data:
{"dxsEntries":[{"dxsId":251659266,"value":126827689936065350532352442368.000000},{"dxsId":251659278,"value":100.000000},{"dxsId":251658754,"value":12572.312500},{"dxsId":251659010,"value":139431858794556274245632.000000},{"dxsId":251659279,"value":100.000000}],"session":{"sessionId":0,"roleId":0},"status":{"code":0}}

kostal-piko-ba.0
2023-02-16 18:42:58.759 debug Piko-BA live data 2 update - Kostal response
data:
{"dxsEntries":[{"dxsId":67109377,"value":0.000000},{"dxsId":67109378,"value":0.000000},{"dxsId":67109379,"value":0.000000},{"dxsId":67109633,"value":0.000000},{"dxsId":67109634,"value":0.000000},{"dxsId":67109635,"value":0.000000},{"dxsId":67109889,"value":0.000000},{"dxsId":67109890,"value":0.000000},{"dxsId":67109891,"value":0.000000},{"dxsId":83887106,"value":0.000000},{"dxsId":83887362,"value":0.000000},{"dxsId":83887618,"value":0.000000}],"session":{"sessionId":0,"roleId":0},"status":{"code":0}}

kostal-piko-ba.0
2023-02-16 18:42:58.757 debug Piko-BA live data 1 update - Kostal response
data:
{"dxsEntries":[{"dxsId":33556736,"value":0.000000},{"dxsId":67109120,"value":0.000000},{"dxsId":33555203,"value":0.000000},{"dxsId":33555201,"value":0.000000},{"dxsId":33555202,"value":0.000000},{"dxsId":33555459,"value":0.000000},{"dxsId":33555457,"value":0.000000},{"dxsId":33555458,"value":0.000000},{"dxsId":33555715,"value":0.000000},{"dxsId":33555713,"value":0.000000},{"dxsId":33555714,"value":0.000000},{"dxsId":83888128,"value":0.000000},{"dxsId":83887872,"value":0.000000},{"dxsId":16780032,"value":0},{"dxsId":33556226,"value":0.000000},{"dxsId":33556227,"value":0.000000},{"dxsId":33556229,"value":0.000000},{"dxsId":33556238,"value":0.000000},{"dxsId":33556230,"value":0}],"session":{"sessionId":0,"roleId":0},"status":{"code":1}}

kostal-piko-ba.0
2023-02-16 18:42:48.757 debug Piko-BA live data 2 update - Kostal response
data:
{"dxsEntries":[{"dxsId":67109377,"value":0.000000},{"dxsId":67109378,"value":0.000000},{"dxsId":67109379,"value":0.000000},{"dxsId":67109633,"value":0.000000},{"dxsId":67109634,"value":0.000000},{"dxsId":67109635,"value":0.000000},{"dxsId":67109889,"value":0.000000},{"dxsId":67109890,"value":0.000000},{"dxsId":67109891,"value":0.000000},{"dxsId":83887106,"value":0.000000},{"dxsId":83887362,"value":0.000000},{"dxsId":83887618,"value":0.000000}],"session":{"sessionId":0,"roleId":0},"status":{"code":0}}

kostal-piko-ba.0
2023-02-16 18:42:48.755 debug Piko-BA live data 1 update - Kostal response
data:
{"dxsEntries":[{"dxsId":33556736,"value":0.000000},{"dxsId":67109120,"value":0.000000},{"dxsId":33555203,"value":0.000000},{"dxsId":33555201,"value":0.000000},{"dxsId":33555202,"value":0.000000},{"dxsId":33555459,"value":0.000000},{"dxsId":33555457,"value":0.000000},{"dxsId":33555458,"value":0.000000},{"dxsId":33555715,"value":0.000000},{"dxsId":33555713,"value":0.000000},{"dxsId":33555714,"value":0.000000},{"dxsId":83888128,"value":0.000000},{"dxsId":83887872,"value":0.000000},{"dxsId":16780032,"value":0},{"dxsId":33556226,"value":0.000000},{"dxsId":33556227,"value":0.000000},{"dxsId":33556229,"value":0.000000},{"dxsId":33556238,"value":0.000000},{"dxsId":33556230,"value":0}],"session":{"sessionId":0,"roleId":0},"status":{"code":1}}

admin.0
2023-02-16 18:42:39.177 info <== Disconnect system.user.admin from
::ffff:192.168.193.90 echarts

admin.0
2023-02-16 18:42:39.176 info <== Disconnect system.user.admin from
::ffff:192.168.193.90 echarts-show

admin.0
2023-02-16 18:42:39.177 info <== Disconnect system.user.admin from
::ffff:192.168.193.90 echarts

admin.0
2023-02-16 18:42:39.176 info <== Disconnect system.user.admin from
::ffff:192.168.193.90 echarts-show

kostal-piko-ba.0
2023-02-16 18:42:38.757 debug Piko-BA live data 2 update - Kostal response
data:
[... cut ...]
{"dxsEntries":[{"dxsId":67109377,"value":0.000000},{"dxsId":67109378,"value":0.000000},{"dxsId":67109379,"value":0.000000},{"dxsId":67109633,"value":0.000000},{"dxsId":67109634,"value":0.000000},{"dxsId":67109635,"value":0.000000},{"dxsId":67109889,"value":0.000000},{"dxsId":67109890,"value":0.000000},{"dxsId":67109891,"value":0.000000},{"dxsId":83887106,"value":0.000000},{"dxsId":83887362,"value":0.000000},{"dxsId":83887618,"value":0.000000}],"session":{"sessionId":0,"roleId":0},"status":{"code":0}

hombach, 2023-02-16:
Perfekt - dann noch den Debug Log wieder aus und Du solltest Ruhe im Log haben - Danke für die geduldige Mitarbeit!

closed 2023-02-20 by hombach