hombach/ioBroker.kostal-piko-ba #583: PIKO 5.0-2 MP plus liefert niocht alle Werte

Chrishclum, 2024-03-21:
Hallo Hombach, ich besitze einen PIKO 5.0-2 MP plus den ich letztes Jahr installiert habe. Nun habe ich den Kostal Piko BA Adapter installiert. Leider liefert er mir nicht alle, für mich interessanten Werte, wie z.B. Tagesproduktion, aktuelle Produktion etc. Das sollte dieser Adapter doch eigentlich können denke ich. Was mache ich falsch? Selbstverständlich habe ich die IP Adresse eingeben, Er zeigt mir auch den total yield, aber schöner wäre es den total yield of the day zu bekommen.

hombach, 2024-03-21:
Bei den MP Plus kommt nach dem was ich bisher herausfinden konnte nur der total yield über die Schnittstelle. Die älteren Geräte wie die Piko BA haben eine ganz andere Schnittstelle mit anderen Daten.

hombach, 2024-03-21:
Aber du kannst mal in deinem EDGE, Chrome oder so eingeben:  http://IP-ADRESSE-DES-PIKO/yields.xml  Dann setz das mal hier rein.... evtl. hat sich da etwas getan in letzter Zeit.

Chrishclum, 2024-03-21:
<root>
<Device Name="PIKO 5.0-2 MP plus" Type="Inverter" Platform="Net16" HmiPlatform="HMI17" NominalPower="4600" UserPowerLimit="nan" CountryPowerLimit="4600" Serial="meine Seriennummer" OEMSerial="meineNummer" BusAddress="1" NetBiosName="INV008882670045" WebPortal="PIKO Solar Portal" ManufacturerURL="kostal-solar-electric.com" IpAddress="192.168.178.47" DateTime="2024-03-21T22:19:15" MilliSeconds="206">
<Yields>
<Yield Type="Produced" Slot="Total" Unit="Wh">
<YieldValue Value="3452960" TimeStamp="2023-05-25T15:30:00"/>
</Yield>
</Yields>
</Device>
</root>

Chrishclum, 2024-03-21:
Anbei

<root>
<Device Name="PIKO 5.0-2 MP plus" Type="Inverter" Platform="Net16" HmiPlatform="HMI17" NominalPower="4600" UserPowerLimit="nan" CountryPowerLimit="4600" Serial="meineserial" OEMSerial="meineserial" BusAddress="1" NetBiosName="INV008882670045" WebPortal="PIKO Solar Portal" ManufacturerURL="kostal-solar-electric.com" IpAddress="192.168.178.47" DateTime="2024-03-21T22:19:15" MilliSeconds="206">
<Yields>
<Yield Type="Produced" Slot="Total" Unit="Wh">
<YieldValue Value="3452960" TimeStamp="2023-05-25T15:30:00"/>
</Yield>
</Yields>
</Device>
</root>

Gruß

closed 2024-03-21 by Chrishclum

Chrishclum, 2024-03-24:
Irgendetwas aufälliges oben zu sehen? Adaptermäßig sollte es doch möglich sein die Wewrte auszulesen

hombach, 2024-03-24:
Wie Du siehst.... 
<Yield Type="Produced" Slot="Total" Unit="Wh">
<YieldValue Value="3452960" TimeStamp="2023-05-25T15:30:00"/>
es gibt nur "Produced Total"