hombach/ioBroker.kostal-piko-ba #589: Warnung/Störung - Verifiy IP-Adress, Connection Timeout Error

patricknitsch, 2024-04-12:
**Describe the bug**  
Ich erhalte immer wieder folgende zwei Meldungen; erst als Warnung, danach als Error:
```
kostal-piko-ba.0
2024-04-12 19:56:42.529	warn	Please verify the IP address: 192.168.30.20 !! (eBA1.2)

kostal-piko-ba.0
2024-04-12 19:56:42.526	warn	Connection timeout error when calling Piko(-BA) API for live data
```

**To Reproduce**  
Steps to reproduce the behavior:
Der Fehler kommt sporadisch. Probleme beim auslesen gibt es nicht. Die beiden Fehler tauchen immer zusammen auf.

**Expected behavior**  
Was ist der Grund der Meldungen und wie kann ich das Problem lösen?

**Screenshots & Logfiles**  

```
2024-04-12 19:56:42.526 - warn: kostal-piko-ba.0 (369) Connection timeout error when calling Piko(-BA) API for live data
--
2024-04-12 19:56:42.529 - warn: kostal-piko-ba.0 (369) Please verify the IP address: 192.168.30.20 !! (eBA1.2)
```

Zeit zwischen Datenabfragen Wechselrichter: 5000 ms
Zeit zwischen Datenabfragen tägliche Statistik: 60000 ms
Zeit zwischen Datenabfragen Allzeitstatistik: 200000 ms

**Versions:**  
 - Adapter version: 3.1.0
 - JS-Controller version: 5.0.19
 - Node version: 18.17.1
 - Operating system: Proxmox; LXC Container mit Debian Bullseye

**Additional context**  
Add any other context about the problem here.

hombach, 2024-04-21:
Wenn die Fehler ungleichmäßig sporadisch auftreten, dann handelt es sich lediglich um ein timeout des Kostal beim Antworten auf die Anfragen. Kommt immer wieder mal vor, die Hardware im Kostal ist nicht so irre potent was die Netzwerk-Kommunikation betrifft.  Wenn sonst alles funktioniert, dann brauchst du dir da keine Kopf machen,
Aber ich will mal versuchen die Meldung von Fehlern etwas zu verringern für solche Fälle.

hombach, 2024-04-21:
Änderungen in 4.0.0 enthalten - Test sehr willkommen ;)

patricknitsch, 2024-04-22:
Hab's installiert und werde es beobachten.

Danke erstmal für deine Unterstützung!!

patricknitsch, 2024-05-06:
Sieht bis jetzt gut aus. Danke für die schnelle Änderung!

closed 2024-05-06 by patricknitsch