hombach/ioBroker.kostal-piko-ba #632: Update stable version in repo from 4.0.0 to 4.0.1

GermanBluefox, 2024-07-21:
# Think about update stable version to 4.0.1
**Version**: stable=**4.0.0** (90 days old) => latest=**4.0.1** (26 days old)
**Installs**: stable=**162** (37.16%), latest=**22** (5.05%), total=**436**

Click to use [developer portal](https://www.iobroker.dev/adapter/kostal-piko-ba/releases)
Click to [edit](https://github.com/ioBroker/ioBroker.repositories/edit/master/sources-dist-stable.json#L1289)

**Do not close this issue manually as a new issue will be created if condition for update still exists.**

Please drop a comment if any reason exists which blocks updating to version 4.0.1 at this time.

Note: This is an automatically generated message and not personally authored by bluefox!
      @mcm1957 for evidence

hombach, 2024-07-21:
mission accomplished ;)

hombach, 2024-07-21:
DONE

closed 2024-07-21 by hombach